OnionUI/Onion #1702: Japanese language cut off in header & footer

jacola, 2024-11-14:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.3.1-1

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Labels rendered in `src/common/theme/render/header.h` and `src/common/theme/render/footer.h` get cut off, while labels in `src/common/theme/render/list.h` are displayed correctly. This is in the default Silky theme which uses a font that supports Japanese language and doesn't use rendered images for the footer text.

For example:
* The list it says `言語を変更` (change language), and the header reads `言語を` (particle + language) when selected. 
* On the bottom, it says `(A)選` when the text in the translation says `選択`
* On the bottom, it says `(B)る` when the text in the translation says `戻る`

[image]
[image]

I believe most of these kanji characters are 2 bytes long, while representing 1 character. I wonder if it's only rendering the number of bytes using character length assuming 1 byte/character. However, for some reason, Settings (`設定`) displays correctly.

### Steps to Reproduce

* Go to `Settings` →`Change Language` → `日本語（Japanese）`
* Press `(A) Select`
* Look at `言語を変更` (change language) in the list view, press `(A) 選` (this is wrong) and see `言語を` at the top missing `変更`

### Provide any additional context or information that may help us investigate your issue

I validated that the format of the translation files and they appear correct.
```
~/devel/Onion/static/build/miyoo/app/lang (main) » file -I ja.lang
ja.lang: application/json; charset=utf-8
~/devel/Onion/static/build/miyoo/app/lang (main) » file -I ja.lang.short
ja.lang.short: application/json; charset=utf-8
```

jacola, 2024-11-14:
It definitely has to do with the mix of hiragana and kanji (or something Japanese specific). When I choose Simplified Chinese, it can display two characters on the footer.
[image]

fengfrw, 2024-11-14:
It is probably necessary to change the font of topic references to support perfect characters.

jacola, 2024-11-14:
I tested with Noto Sans (added the font and references) and the result was the same. The list, header and footer all use the same font too, so I think it's unrelated to the font.

fengfrw, 2024-11-14:
This is an example of gruvbox font.
[image]
[image]

jacola, 2024-11-14:
Thanks. I'll try modding the default theme's font or using that one.

jacola, 2024-11-14:
Cool, if the `config.json` file is updated to use `HENB.ttf` instead of the `Exo-2-Bold-Italic_Universal.ttf` then it does work. Thanks again @fengfrw 🙇 
[image]

It'd be better if the default theme worked with all languages, but at least there's a workaround.

closed 2024-11-14 by jacola

jacola, 2024-11-14:
[SilkyJapanese by jacola.zip](https://github.com/user-attachments/files/17744557/SilkyJapanese.by.jacola.zip)