OnionUI/Onion #1703: Issue resuming and time tracking games with the same name

jsteel44, 2024-11-15:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.1-1

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

For example, if a game on the Mega Drive and Game Gear have the same name, eg "Sonic.bin" and "Sonic.gg", both appear in the game switcher as "Sonic". I think this is causing confusion with time tracking and auto-resuming via save states.

Time tracking on these games (with duplicate names) report 0s and never advance.

Save states fail to resume if you switch from one game to the other. I suspect it is trying to resume the save state for the other game and the game freezes/crashes. If you close and load the same game, the save resumes fine, switching between the two does not work.

A solution I have used is to find these duplicates and rename them, then the issue is gone. But I think this should be fixed, for example it could show "Sonic (MD)" and "Sonic (GG)", or show the file extension, and as long as the time tracker and "resuming from save state" know about this, it should avoid the conflict.

### Steps to Reproduce

1. Load two games with the same name (different file extension) (case sensitive) eg Sonic.md and Sonic.gg
2. With both in the game switcher, try switching between the two and notice the auto save state resume crash the game
3. Try playing both games and notice that both (or at least one) fails to progress the time displayed in the game switcher (always 0s)

### Provide any additional context or information that may help us investigate your issue

Remember the games need to have an identical name (not file extension) to trigger the issues. For example Sonic.md and sonic.gg would not be an issue, only if they share the same name of Sonic.md and Sonic.gg.

jsteel44, 2024-11-16:
I was playing around with this some more after setting up a new SD card I couldn't get the time tracker issue to trigger, but the save state issue switching between two games with the same name is still there. I'll try and find a better way to trigger the time tracker issue.

To clarify, I'm not trying to use names with the same name, actually the file names are wildly different but the game switcher strips out a lot of the text, eg "Sonic (Europe).md" and "Sonic (USA).gg" become simply "Sonic" in the game switcher. I'd be happy to keep the full name to avoid the conflict, but I'm unsure how.

QuackWalks, 2025-04-23:
With the latest V4.4.0-beta2, RetroArch now has options to sort and write saves by content directory instead of the core directory. This solves the issue for loading games but not the problem with activity tracker.