OnionUI/Onion #1707: Issue Title

megusta420, 2024-11-25:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Emulated time skip without RTC - how does it work?

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Hey folks!
I set up Tweaks->System->Date and time->Emulated time skip to be 4 hours, so I expect it to add +4 hours after each reboot of Miyoo. Should it work that way or am I missing something?
Because after each reboot the system adds +3 hours from 00:00, but not 00:00 to 04:00 to 08:00, etc.
So each reboot my date is 1970-01-01 03:00am.
Mostly play Pokemon Unbound so it's quite crucial for me.
Onion OS 4.3.1-1

### Steps to Reproduce

Set up Tweaks->System->Date and time->Emulated time skip to be 4 hours

### Provide any additional context or information that may help us investigate your issue

No additional context

Aemiii91, 2024-11-25:
Should be in `.tmp_update/`, but it's something you would've added manually

megusta420, 2024-11-25:
> Should be in `.tmp_update/`, but it's something you would've added manually

Yup, renamed that file and now works as expected! Thanks to you all guys! Didn't create that file previously, but now all is good and works as expected!

closed 2024-11-25 by megusta420