OnionUI/Onion #1710: Battery Uncalibrated, No way to enable AXP223 calibration mode.

DevL0rd, 2024-12-02:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.1-1

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

My understanding is that onion os is getting battery percent readings directly from the AXP223 chippies.

Unfortunately it's the most in-accurate reading I have ever seen.

Additionally after reading over the document for this controller, there is in fact a calibration mode that can be set.

I believe calibrating it may resolve the issue with crazy battery readings.

May there possibly be a way to toggle the batteries calibration mode from the advanced tools?

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

Use the onion os.
Cry at battery reading inaccuracies.

tGecko, 2024-12-03:
Is this on the Miyoo Mini or Miyoo Mini plus?

DevL0rd, 2024-12-03:
> Is this on the Miyoo Mini or Miyoo Mini plus? Only the plus uses the AXP223, the regular mini has no such chip

Oh really? Interesting.

Perhaps I can correct my own voltage curve to something more linear.

Can you give me some guidance as to how the percentage is calculated at the moment and where in code that may be?

I assume cause of the mostly constant voltage tho that may be impossible but I'd like to experiment.

tGecko, 2024-12-03:
Basically on the Miyoo Mini, all we have is the battery voltage. it is read by batmon [here](https://github.com/OnionUI/Onion/blob/58916b8bd3308853918991e5bfa25911200a72e0/src/batmon/batmon.c#L84) and then written to /tmp/.axp_result [here](https://github.com/OnionUI/Onion/blob/58916b8bd3308853918991e5bfa25911200a72e0/src/batmon/batmon.c#L325)

MainUI (the closed source frontend) reads the value from that file and displays it accordingly.