OnionUI/Onion #1722: RTC Not Functioning on 4.4.0 beta

Ziko577, 2024-12-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-8d8dc7ed

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

RTC is no longer fucntioning as intended as of this release and now when I boot up Emerald Seaglass, I am getting an internal battery error message instead of the game booting and showing the current time as it was set to do in the stable v4.3.1-1. I used the .NoTimeRestore file before this and it was setup correctly with no issue and the new option in the gpsp core was set to automatic detection with no changes. 

### Steps to Reproduce

Check to see if the .NoTimeRestore file is there and also check to see under mGBA and gpsp if RTC is set to auto and check to see if current time matches up in game to the time on the device.

### Provide any additional context or information that may help us investigate your issue

[.noTimeRestore.zip](https://github.com/user-attachments/files/18224383/default.noTimeRestore.zip)

tGecko, 2024-12-23:
Can you make sure you are running the game using mGBA? In game open the retroarch menu using Menu+Select, then go to Information and look for the core name.

In the last update, users were prompted to transfer to gpSP if a save already existed, or were automatically set to use gpSP if not.

Ziko577, 2024-12-23:
The game was running under gpsp the whole time. I never moved it to mGBA but I noticed that the save file wasn't there and had to be renamed to .srm as it was .sav before. It didn't resolve the issue and instead it gave the dry battery error. Someone else noted that by deleting the .NoTimeRestore file may have fixed the issue as I have it under there. I can try and see if that will help.

Aemiii91, 2024-12-23:
Closing this as duplicate of #1723 

(I know this was created first)

Aemiii91, 2024-12-23:
Duplicate of #1723

closed 2024-12-23 by Aemiii91