OnionUI/Onion #1723: Issue Title

LuckyGray7, 2024-12-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.4.0-8d8dc7ed

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

This update appears to have broken RTC for me in Pokemon Unbound, it is telling me to turn on RTC but I can't find the option anywhere and my MM+ is the newer one with RTC built in. Anyone else having this issue?

### Steps to Reproduce

update to v4.4.0-8d8dc7ed

### Provide any additional context or information that may help us investigate your issue

[image]
[image]

Aemiii91, 2024-12-23:
Did you switch to gpSP or are you still playing on mGBA?

LuckyGray7, 2024-12-23:
I swapped over to gpSP and it appears I have corrected this by deleting .noTimeRestore

However it appears none of my saves convert over to gpSP. trying to continue with Pokemon Unbound and it just causing me to start a new game even if I manually move save states and saves over. Even if I rename them .srm or .sav

Aemiii91, 2024-12-23:
Yes, all evidence seems to indicate that saves are not transferable, and we'll probably remove that option.

Aemiii91, 2024-12-23:
Does your device have RTC, or did the .noTimeRestore appear by itself?

LuckyGray7, 2024-12-23:
my Device does have RTC built in, it's one of the newer ones. I had placed the .noTimeRestore file in myself to prevent the TimeRestore function of the Clock App from overwriting the time so that the RTC can be in charge of it. However it seems this file is causing RTC to break with the new update. Removing the file seems to restore RTC functionality as intended with the new beta update.

LuckyGray7, 2024-12-23:
> my Device does have RTC built in, it's one of the newer ones. I had placed the .noTimeRestore file in myself to prevent the TimeRestore function of the Clock App from overwriting the time so that [...]

my Device does have RTC built in, it's one of the newer ones. I had placed the .noTimeRestore file in myself to prevent the TimeRestore function of the Clock App from overwriting the time so that the RTC can be in charge of it. However it seems this file is causing RTC to break with the new update. Removing the file seems to restore RTC functionality as intended with the new beta update.

tGecko, 2024-12-23:
This is weird, the .noTimeRestore file will appear automatically if your device has RTC, and should not have any impact whatsoever on the emulators RTC capablitiy. I can confirm that RTC is an issue here as well, but deleting the file does not have any effect on that for me (as expected)

LuckyGray7, 2024-12-23:
> This is weird. The .noTimeRestore file will appear automatically if your device has RTC, and should not have any impact whatsoever on the emulators RTC capablitiy. I can confirm that RTC with gpSP is an issue here as well, but deleting the file does not have any effect on that for me (as expected)

Actually I think changing the core option of RTC from auto to on, and then restarting the game fixed it. because the no RTC warning went away I think I just did that and deleted the file which made me think it was the file. I'm gonna restore the file and see if I can keep running it with RTC active and the .notimerestore file .

LuckyGray7, 2024-12-23:
yeah it looks like the .noTimeRestore automatically came back.

Try hitting Select+Menu going to core options, changing RTC support from Auto to On and then restart, close content and then launch the game again. 

That appears to be what fixed RTC for me, I think Auto was not turning it on correctly.

tGecko, 2024-12-23:
Turns out the issue is that gpSP uses an internal database on when to automatically enable RTC, and this database does not contain romhacks.

**For these games to work properly:** Force enable RTC in RetroArch menu > Core Options, then from MainUI select the game, press Y, Reset game. RTC should now work.

closed 2024-12-23 by tGecko