OnionUI/Onion #1724: Game Switcher Broken in v4.4.0

mrtekkid, 2024-12-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-8d8dc7ed

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

When playing a game (in my instance gameboy advance game) pressing the menu button to trigger game switcher does not work. Also power button is unresponsive.

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

When pressing the menu button Miyoo will buzz like it’s activating game switcher but does not display. Need to press select + menu to trigger retro arch menu and exit through there. Uncertain if saving in this state currently. Also the power button is non-functional until quit from the retroarch menu.

tGecko, 2024-12-23:
Please do the following:

-  enable logging in Apps > Tweaks > Advanced > Diagnostics
- replicate the issues you have
- create a log snapshot from Apps > Tweaks > Advanced > Diagnostics
- upload the log_export.7z from the root of your SD card here

jlnr, 2024-12-23:
I've had the switcher button break in 4.3.1 too, but could never figure out how to reliably reproduce it.

tomahawk-k, 2024-12-23:
I have the same issue. Can only exit games by opening the retroarch menu
[log_export.zip](https://github.com/user-attachments/files/18227463/log_export.zip)

Aemiii91, 2024-12-23:
@tomahawk-k thanks! Can you try resetting RA setting via **Apps -> Tweaks -> Advanced -> Reset settings** ?

tomahawk-k, 2024-12-23:
@Aemiii91 Thank you! Resetting the main configuration worked and the game switcher is now working

Aemiii91, 2024-12-23:
Alright, workaround confirmed. We're working on a permanent fix ✅

devcodable, 2024-12-23:
Hello, reporting back from reddit with the logs. Additional info that may be interesting:

- When in game if I do an action involving power/menu buttons it only takes place once I quit retroarch (like power off or change game with double press)
- This is reproducible both in MM+ and MMv4, both in the latest firmwares for each one.
- Tried gba, gbc and arcade, it's reproducible in all of them

Thank you for the amazing work, if you need more debugging on my side please let me know.

[log_export.zip](https://github.com/user-attachments/files/18228271/log_export.zip)

tGecko, 2024-12-23:
> ###
> **[devcodable](/devcodable) ** commented [Dec 23, 2024](#issuecomment-2559387689) •

Hey, thanks for your logs, please on your device in RetroArch/.retroarch/retroarch.cfg edit `network_cmd_enable = "false"` to `network_cmd_enable = "true"` then try again

Aemiii91, 2024-12-23:
Fixed in 51bde3716341c07768e96a00686aaa129ba97b57

closed 2024-12-23 by Aemiii91