OnionUI/Onion #1725: Migrating Game from Old Core to New Core Fails

mrtekkid, 2024-12-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

Migration does not work to new core

### Steps to Reproduce

When playing Minsh Cap after upgrading promoted to move save to new core. Said yes. Switched cores and was promoted to create a new game. 

Had to change to old core to continue playing the game. 

### Provide any additional context or information that may help us investigate your issue

Uncertain if there is a way to re-trigger this or manually do it.

Aemiii91, 2024-12-23:
This will need more testing. We can't be sure saves are even transferrable anymore.

Aemiii91, 2024-12-23:
If none of people's saves can be transferred, we might just need to remove the option.

Aemiii91, 2024-12-23:
The transfer option has been removed until gpSP supports it (6cd13b4462c3833b4780543afc611e5e0a89c096)

closed 2024-12-23 by Aemiii91