OnionUI/Onion #1742: Problem with audio on FBneo 2012 when using the GameSwitcher

vilaron, 2025-01-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.4.0-beta

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When I switch from a Neo geo Game to any other game, and then return to the Neo Geo game, the audio no longer works and I have to restart the game using the retroarch menu in order to restore the audio. I have the neogeo.zip on the BIOS and roms/neogeo folder.

### Steps to Reproduce

Play any neo game -> go to the GameSwitcher -> Choose any game -> go again to the GameSwitcher and return to the original neo geo game.

### Provide any additional context or information that may help us investigate your issue

I am using the "Neo-Geo Rom Collection By Ghostware" ROM set

QuackWalks, 2025-01-29:
I try to reproduce but sound works fine for me.

vilaron, 2025-01-29:
Update: I reset everthing in the Reset settings and that solve the issue. Of course, I had to configure all of my settings again. I do not know what cause the problem, but in the end there was a solution. Thanks a lot for the help.