OnionUI/Onion #1748: Isn't possible to upgrade to the latest beta

Amiga500, 2025-01-15:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-3ba4456d

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

Isn't possible to upgrade to the latest beta. Please check the pic attached for more info

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

[image]

Aemiii91, 2025-01-15:
Thanks for the report, I'll fix it asap 👌

Aemiii91, 2025-01-15:
@Amiga500 can you try again?

miguelbaldi, 2025-01-15:
I've tried now and the problem seems to persist 
[image]

Amiga500, 2025-01-15:
> @Amiga500 can you try again?

Tried now. Same screen as miguelbaldi

Aemiii91, 2025-01-15:
Should be good now

Amiga500, 2025-01-16:
mmmhhh no :(

Amiga500, 2025-01-16:
Very strange! Can i go with a manual update, or you want time to think for a solution in order to test for it?

Aemiii91, 2025-01-16:
No no, go ahead and do a manual update 👍

Amiga500, 2025-01-16:
Closed as mystery of faith