OnionUI/Onion #1753: Long press to exit takes longer and savestate issue

JordonVW, 2025-01-17:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.4.0-9A3687ED

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

In my tweaks settings I have long press in game set to exit, it used to take just 2 or 3 settings to exit. It now takes twice as long and wometimes doesnt exit until i let go of the button.

 It appears to try save a savestate on exit and then load one the next time I go to this same game. The game is using the mgba core and an image of a screenshot from a savestate shows when loading. 

It doesn't load the savestate shown in the image as it doesn't seem to be a real saved state so it's just a delayed process on exit and loading too.

### Steps to Reproduce

Use long press to exit a game with mgba core.

### Provide any additional context or information that may help us investigate your issue

I cannot see a savestate setting anywhere that is trying to override what I have had set up as in the past.

Aemiii91, 2025-01-17:
RetroArch startup and exit to menu hasn't  changed in this release. Yes, the save screenshot is shown while loading, but doesn't affect the time it takes.

Is it the same slow exit experience if you press MENU+SELECT, B, and select "Quit RetroArch"?

At a last resort, try resetting all RetroArch related settings via **Tweaks -> Advanced -> Reset settings**.

Aemiii91, 2025-01-17:
Furthermore, if you've disabled "auto load state", then it won't load the save state. And if you've disabled "auto save state on exit", then the saving on exit won't happen.

JordonVW, 2025-01-17:
The savestate screenshot seems to be there by default now because even games I haven't played yet and dont have a savestate are searching for one when opening the game.

Yes quiting the game through the retroarch menu works normally. 

I definitely haven't tried to enable anything to do with savestates since doing the update.

Aemiii91, 2025-01-17:
Could you attach a copy of your `RetroArch/.retroarch/retroarch.cfg`?

JordonVW, 2025-01-17:
> Furthermore, if you've disabled "auto load state", then it won't load the save state. And if you've disabled "auto save state on exit", then the saving on exit won't happen.

I must have disabled auto save and load states that long ago I forgot that was the default setting. Yeah they're turned off still despite it visually showing it trying to show a screenshot of a savestate when there isn't one to load.

JordonVW, 2025-01-17:
> What do you mean by "searching for one"?
> 
> Auto load and save states are enabled by default in Onion, these would only be disabled if you've overwritten the default configuration.

When the game doesn't have a screenshot it showed a blank screen where it had a screenshot for the game i had played already then it gave up and loaded normally.

JordonVW, 2025-01-17:
> Could you attach a copy of your `RetroArch/.retroarch/retroarch.cfg`?

Yeah sure I'll get my laptop again and get it now.

JordonVW, 2025-01-17:
> Could you attach a copy of your `RetroArch/.retroarch/retroarch.cfg`?

It says i cannot upload a .cfg file, so this is the contents if the file.

accessibility_enable = "false"
accessibility_narrator_speech_speed = "5"
ai_service_enable = "false"
ai_service_mode = "1"
ai_service_pause = "false"
ai_service_source_lang = "0"
ai_service_target_lang = "0"
ai_service_url = "http://localhost:4404/"
all_users_control_menu = "false"
apply_cheats_after_load = "false"
apply_cheats_after_toggle = "false"
aspect_ratio_index = "22"
assets_directory = ":/.retroarch/assets"
audio_block_frames = "0"
audio_device = ""
audio_driver = "oss"
audio_dsp_plugin = ""
audio_enable = "true"
audio_enable_menu = "false"
audio_enable_menu_bgm = "false"
audio_enable_menu_cancel = "false"
audio_enable_menu_notice = "false"
audio_enable_menu_ok = "false"
audio_enable_menu_scroll = "false"
audio_fastforward_mute = "false"
audio_filter_dir = ":/.retroarch/filters/audio"
audio_latency = "64"
audio_max_timing_skew = "0.050000"
audio_mute_enable = "false"
audio_out_rate = "48000"
audio_rate_control = "true"
audio_rate_control_delta = "0.005000"
audio_resampler = "nearest"
audio_resampler_quality = "2"
audio_sync = "true"
audio_volume = "0.000000"
auto_overrides_enable = "true"
auto_remaps_enable = "true"
auto_screenshot_filename = "true"
auto_shaders_enable = "true"
autosave_interval = "0"
block_sram_overwrite = "false"
bluetooth_driver = "null"
builtin_imageviewer_enable = "false"
builtin_mediaplayer_enable = "false"
bundle_assets_dst_path = ""
bundle_assets_dst_path_subdir = ""
bundle_assets_extract_enable = "false"
bundle_assets_extract_last_version = "0"
bundle_assets_extract_version_current = "0"
bundle_assets_src_path = ""
cache_directory = ""
camera_allow = "false"
camera_device = ""
camera_driver = "null"
cheat_database_path = ":/.retroarch/cheats"
check_firmware_before_loading = "false"
cheevos_appearance_anchor = "0"
cheevos_appearance_padding_auto = "true"
cheevos_appearance_padding_h = "0.000000"
cheevos_appearance_padding_v = "0.000000"
cheevos_auto_screenshot = "false"
cheevos_badges_enable = "false"
cheevos_challenge_indicators = "true"
cheevos_custom_host = ""
cheevos_enable = "false"
[... 3087 lines cut ...]
video_stream_port = "56400"
video_stream_quality = "11"
video_stream_scale_factor = "1"
video_stream_url = ""
video_swap_interval = "1"
video_threaded = "false"
video_vsync = "false"
video_waitable_swapchains = "true"
video_window_auto_height_max = "1080"
video_window_auto_width_max = "1920"
video_window_custom_size_enable = "false"
video_window_opacity = "100"
video_window_save_positions = "false"
video_window_show_decorations = "true"
video_windowed_fullscreen = "true"
video_windowed_position_height = "720"
video_windowed_position_width = "1280"
video_windowed_position_x = "0"
video_windowed_position_y = "0"
vrr_runloop_enable = "false"
wifi_driver = "null"
wifi_enabled = "true"
xmb_font = ""
youtube_stream_key = ""
network_cmd_enable = "true"

Aemiii91, 2025-01-18:
Try the latest beta, it should handle this use much better.

- we will no longer display the saving message if "auto save state" is OFF
- I've confirmed that this yields the same "time to exit" in mGBA as in v4.3

JordonVW, 2025-01-18:
> Try the latest beta, it should handle this use case much better.
> 
> * we will no longer display the saving message if "auto save state" is OFF
> * I've confirmed that this yields the same "time to exit" in mGBA as in v4.3

Yeah it no longer shows it on exit so that has worked. Not sure what had changed but it hadn't showed these save state load or saves for the last year or so I've had the save states off.

It still shows it is trying to load a save state upon opening a game though whether that's something you can disable if the user doesn't use the auto load state setting.

It seems the functionality of the long press of the hotkey button on my device has become unreliable which is why it felt like it was taking longer to exit.

It wasn't exiting as fast like it used to, to work around it I've changed exit to double press and it works fine with that enabled instead.

Aemiii91, 2025-01-18:
Loading just means "loading the game" in this case. I'm not sure what would cause the long press to work slower than double press, except you need to hold it for 300ms, but it seems isolated. Closing the issue.

closed 2025-01-18 by Aemiii91