OnionUI/Onion #1757: RTC Still Not Working Under gpsp Core

Ziko577, 2025-01-20:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.4.0-37449678`

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

After the latest beta, the default gpsp core still isn't working with RTC despite it set to being detected automatically plus having the .NoTimeRestore file in the config. To alleviate this, I used the GLO to set Pokemon Modern Emerald which is a romhack specifically for mGBA back to it as I kept the folder when it was updated. Pokemon Emerald Seaglass also presents this problem as well. 

### Steps to Reproduce

Booting an RTC game, with the setting set to automatic instead of ON should cause the error to pop up in a game that supports it. 

### Provide any additional context or information that may help us investigate your issue

[https://drive.google.com/file/d/1FipzNeOICRgZLnB9Ztw4Fz-srAiLv-Vd/view?usp=drive_link](url)

Aemiii91, 2025-01-20:
We are not the maintaining gpSP. It would  be better to raise this issue at https://github.com/libretro/gpsp

I'll pass this on.

Ziko577, 2025-01-20:
Thank you. I just didn't know where else to put this. Beyond that, keep up
the good work as this thing's much better than before!

radityaargap, 2025-02-24:
> ### Checklist
> * [x]  I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories[x]  I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer[x]  I have searched the existing (open and closed) issues to ensure I do not log a duplicate
> 
> ### Onions OS Version
> Onion-v4.4.0-37449678`
> 
> ### Miyoo Firmware Version
> 202306282128
> 
> ### Provide a clear and concise description of the issue
> After the latest beta, the default gpsp core still isn't working with RTC despite it set to being detected automatically plus having the .NoTimeRestore file in the config folder as a fallback. To alleviate this, I used the GLO to set Pokemon Modern Emerald which is a romhack specifically for mGBA as it messed the sound up under gpsp plus the RTC issue, back to it as I kept the folder when it was updated. I also set an override so that way it boots to mGBA for Modern Emerald specifically. Pokemon Emerald Seaglass also presents this problem as well. I provided a link to a screenshot showing the error message saying that clock-based events will no longer occur.
> 
> ### Steps to Reproduce
> Booting an RTC supported game, with the setting set to automatic instead of ON should cause the error to pop up in a game that supports it.
> 
> ### Provide any additional context or information that may help us investigate your issue
> [image]
> 
> Screenshot taken on OnionOS using a MiyooMini Plus.

Apologize for being OOT, but what shader are you using there?

Ziko577, 2025-02-26:
It's the Nintendo Game Boy Advance one as it's simple and does well for the
low brightness in this thing as it's only on 3/10.