OnionUI/Onion #1758: Game switcher fail

Veuks, 2025-01-20:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-37449678

### Miyoo Firmware Version

202306092122

### Provide a clear and concise description of the issue

The first game change after booting the console with the Game Switcher doesn't work, it goes straight back to the Game Switcher.

Once the bug has occurred once, it doesn't happen again until the next time the console is started.

### Steps to Reproduce

-Switch on the console, which is booting a game
-Press menu button to launch Game Switcher
-Change game
-Press A or menu button to launch new game
-The new game doesn't launch, it returns directly to the Game Switcher

### Provide any additional context or information that may help us investigate your issue

I have a Sandisk SD card, I tested it with several games from several systems.

Veuks, 2025-02-25:
Nobody ?

kilroy01, 2025-04-05:
I noticed a similar issue. I'm on the same 4.4.0 beta version. In my case, it doesn't return to the GameSwitcher, rather it returns to the game I tried to switch from. Seems to happen once after turning the console on but no idea how to reproduce. I can try to get the logs if it helps.

aloven, 2025-08-19:
I've noticed this issue recently a couple times.  I'll try to switch to a new game, GS will think it's switched and even show the new game label, but the actual core / game will be the previous one.  exiting and coming back seems to reset it

Denellyne, 2026-01-17:
I'll try to fix this

RdrSeraphim, 2026-01-18:
> I'll try to fix this

Looks like it might've been fixed by Aemii in [f870aa2](https://github.com/OnionUI/Onion/commit/f870aa26987bcc486559ba28ab73ec52f60ab02a).

closed 2026-01-18 by Aemiii91