OnionUI/Onion #1768: Pressing power button does not suspend the system in 4.4.0

LucasGGamerM, 2025-02-09:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

In 4.3.1, the power button behavior when just tapped was to turn off the screen, inputs, and audio output. 

Now, in 4.4.0, it turns off only the screen and inputs, making it so that the audio continues playing even if the system should be suspended. 

### Steps to Reproduce

Open any game with audio, press the power button, see that the audio continues playing. 

### Provide any additional context or information that may help us investigate your issue

I believe the source of the issue is in commit 1975e9c08674c46459f37581e9acbe05e2790b56, whete there was a change that made it possible to listen to music in GMU with the screen turned off.

Aemiii91, 2025-02-09:
I'm not able to reproduce. Any specific system or game I should try?

LucasGGamerM, 2025-02-10:
I noticed this one when I was playing Chrono Trigger for the SNES, so please try that one.

hqamen, 2025-02-17:
I can confirm that this happens in the latest beta version. It definitely happens in GBA games.

hqamen, 2025-04-03:
When installing from scratch, the problem disappeared. Perhaps this only appears when upgrading from a lower version.

nickterhaar, 2025-04-16:
I can also confirm that the issue only occurs after upgrading from previous version. With a clean install the issues doesn't occur.

LucasGGamerM, 2025-04-18:
> When installing from scratch, the problem disappeared. Perhaps this only appears when upgrading from a lower version.

I see. I will try wiping everything except the saves and media directories and reinstall onion to see what happens

QuackWalks, 2025-04-22:
Updated a v4 as well through the OTA app. The game slows way down, like it's almost done turning down the system, but doesn't quite get all the way there. 

[https://youtube.com/shorts/_AjEjgiqmsI?feature=share](url)

LucasGGamerM, 2025-04-22:
Yeah, it got like that for me too. Thankfully a clean install does indeed fix the issue somehow

QuackWalks, 2025-04-23:
Could everyone who had this problem be sharing something in common, like a certain set of configs from a certain downloaded set of games?

LucasGGamerM, 2025-04-23:
We could diff the files in 4.3 and 4.4 to see what we should look for

QuackWalks, 2025-04-23:
[March backup compared to now.zip](https://github.com/user-attachments/files/19874346/March.backup.compared.to.now.zip)
Here's the difference in files from a month ago. I've deleted and reinstalled the latest beta to fix the problem, but I hope this helps.

Aemiii91, 2026-01-18:
This should be fixed in latest beta

closed 2026-01-18 by Aemiii91