OnionUI/Onion #1779: RetroArch menu interface still using 640x480 on MMv4

CyberWW, 2025-02-26:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.4.0-beta

### Miyoo Firmware Version

202408061133

### Provide a clear and concise description of the issue

when you in game press "menu"+"select" the RetroArch menu just looks not pixelperfect

### Steps to Reproduce

1 open any game
2 press "menu"+"select" open RetroArch menu

### Provide any additional context or information that may help us investigate your issue

[image]

tGecko, 2025-02-27:
The menu we use (rgui) is 320x240 only, it doesn't scale well to 752x560, that's why the font looks wonky. This is the same behavior as for example on PC if you'd use the rgui menu. The alternative would be to have the menu displayed at 640x480, which we decided against because we thought looked even worse.

tGecko, 2025-02-27:
You can disable the stretching and have the menu displayed at 640x480 by creating this empty file:

`SDCARD/.tmp_update/config/RetroArch/.noMenuStretch`