OnionUI/Onion #1782: Game Switcher "Glitch"

SkyWalker541, 2025-03-08:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion 4.4.0-beta

### Miyoo Firmware Version

202310271401

### Provide a clear and concise description of the issue

Game Switcher screen glitches when opening the selected game.

### Steps to Reproduce

Not sure how to reproduce the glitch. I recently started with a new sd card, installed done set 3, and transferred my saves/states to the new install. I did not use the new "sensible arrangement of consoles" provided by done set 3, that adjusts the systems order in emu. That caused issues with configs not loading, so I did not use the "sensible arrangement of consoles".

### Provide any additional context or information that may help us investigate your issue

https://github.com/user-attachments/assets/f883a270-29db-45ab-b6ed-1163de4ec477

Aemiii91, 2025-03-08:
I see this as well, it's a known bug when changing the screen resolution. I think the only way to fix it properly, would be to run all apps in native resolution.

SkyWalker541, 2025-03-08:
> I see this as well, it's a known bug when changing the screen resolution. I think the only way to fix it properly, would be to run all apps in native resolution.

Thank you for the reply 🙏. Apologies if I missed the issue elsewhere.

Aemiii91, 2025-03-08:
You didn't, I just meant it was known to me 😅 it's not really that critical to fix, but of course it's one of our goals to run in native resolution, at some point.

SkyWalker541, 2025-03-08:
All good! I was freaking out for a bit yesterday, thinking I had a corrupt file somewhere while doing a fresh install and transferring all my stuff to the new card. 😩😂 Glad to know it wasn't a corrupt file somewhere. Thanks again!