OnionUI/Onion #1787: Game Switcher leaves uncompressed rom (GBA & GG)

Stevearino42, 2025-03-16:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.4.0.37449678

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When exiting a running GBA or GG game via the Game Switcher, an uncompressed copy of the rom file is left behind in the rom directory, i.e. '/Roms/GBA/Advance Wars (USA) (Rev 1).gba'. The .zip of the rom is there too. This only seems to be happening with GBA and GG roms.

### Steps to Reproduce

Load game from the Games > GBA menu with 'A'. Single press MENU to go to Game Switcher. Press 'B' to exit Game Switcher.

Also happens when I single press MENU to go to Game Switcher, press START, choose Save and press 'A', and then either press 'B' or choose "Exit to menu' to exit the Game Switcher. 

### Provide any additional context or information that may help us investigate your issue

Device is a Miyoo Mini Plus rev 2, purchased about one month ago.

Tested with other cores and emulators - ARCADE, FC, FDS, GB, GBC, MD, MS, NDS, NEOGEO, PCE, PCECD, PS, 
SEGACD, and SFC - but I did not see the uncompressed rom file (for those which use zipped rom files).

If I bypass Game Switcher by setting In-Game Menu > Single Press to Quick Menu, and MainUI Menu > Single Press to Context Menu (in Apps > Tweaks > Button Shortcuts), I do not see the uncompressed rom file when exiting a game with either Menu Single Press > B > Quit RetroArch nor via a Long Press of Menu.

Tried reformatting the SD card and loading v4.3.1-1 but did not see the same problem. Reformatted the SD card again and loaded v4.4.0, and the problem returns.

Note: 'Auto Load' and 'Increment Save State Index Automatically' are turned off globally in RetroArch. I also tried it with them turned back on, but I still get the uncompressed rom file.

[image]

Stevearino42, 2025-03-16:
Forgot to mention, if I 'Refresh all ROMs' after this happens, then I have duplicate entries in the Games lists for any games affected.

brad86m, 2025-04-07:
Currently having the same issue myself, and it's actually causing my OCD to go into overdrive. Spending many hours getting my rom lists all today and organised, only to have duplicate entries show up due to this bug(?).

If you find a solution, I would greatly appreciate the information to stop this from happening.

Stevearino42, 2025-04-07:
> If you find a solution, I would greatly appreciate the information to stop this from happening.

As a work around, I unzipped all of the GBA and GG roms for now.

Shmayro, 2025-10-18:
Still happening on the latest beta