OnionUI/Onion #1791: Battery percent text overlaps wifi icon only in guest mode

slrgt, 2025-03-22:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When in Guest Mode the text of the battery percent overlaps with the Wifi icon

### Steps to Reproduce

enable wifi
enable guest mode

### Provide any additional context or information that may help us investigate your issue

could only test in 4.4 not sure if its a new bug. I believer I put the correct firmware version (the instructions on where to find it from github are different than on the actual device. its in Settings -> About device -> Firmware version)

Aemiii91, 2025-03-23:
Please add some screenshots

slrgt, 2025-03-23:
[image]

Aemiii91, 2025-03-24:
Seems to be a theme fallback, have you tried setting a theme while the guest profile is active?

slrgt, 2025-03-24:
that seemed to have fixed it