OnionUI/Onion #1798: Mgba for gba not avaliable and loss of saves

damo08, 2025-04-15:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.4.0-37449678

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Hi. Now that gpsp is the default emulator you can't select mgba for gba core. I got it working by editing the launch.sh but all my save files and save states have disappeared. I'm not sure if it's because the were converted to gpsp files.

I don't use net play so would it be possible to have a toggle/option to choose between the two cores during setup.

I did try discord but it wasn't working on my phone.

Thanks for all the hard work.

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

Changed on the beta update.

redsky17, 2025-04-18:
I noticed this exact same issue.  From checking the SD card, it appears there's some migration script that runs based on a file that is in the mGBA saves folder (something like `.gpspImportDone` or so), but my saves were not transferred.

There is a workaround, however.  You can go into the package manager and install mGBA core from there.  Once you do that, navigate to the game in your games list and press Y.  This will bring up a menu which will allow you to change which core you are using.  I switched the games I was missing save data from gpsp back to mGBA and the work just fine.  Hope this helps!

damo08, 2025-04-20:
I changed the launch.sh. I had to redo the change on the new beta 2 update. I also backed up my saves first but I didn't lose them this time, it might be because I didn't launch gpsp.

closed 2026-01-20 by Aemiii91