OnionUI/Onion #1803: GameSwitcher is much slower than before

Zazzago, 2025-04-21:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-37449678

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

Compared to the previous version of the GameSwitcher (4.3.1), this new one is generally faster than the old one (GB, gba, snes, arcade), but exiting from PS1 games is much slower than the previous version.

### Steps to Reproduce

Play a PS1 games and exit to GameSwitcher 

### Provide any additional context or information that may help us investigate your issue

loading is slow only when exiting from a ps1 game and not when entering, so there must be some procedure during exit that slows it down (only for PS1 games, other emu Is fast)

Zazzago, 2025-04-21:
Thanks for the res

closed 2025-04-21 by Zazzago

Aemiii91, 2025-04-21:
Great, that sounds more in line with my own testing 👌