OnionUI/Onion #1806: Graphic glitch when open a game from GameSwitcher on MMV4

Zazzago, 2025-04-24:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-37449678

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

From GameSwitcher when open a game, initially, when "loading" should appear, there Is a graphic glitch, as if there was a quick change in resolution (I hypothesize that when opening the game there is a double change in resolution 640x480 and 752x560).
if instead from the game you return to the GameSwitcher, there are no graphic glitches, it only happens on miyoo mini v4 (no plus) from GameSwitcher when loading the game.
then it should be added that GameSwitcher reaches the native resolution of v4 (752x560) only if you enter it while we are in a game, while if you start from the main menu, the GameSwitcher resolution remains 640x480.
I think all these small glitches are due to the fact that Onion OS on v4 initially runs at 640x480 and reaches native resolution only during emulation

### Steps to Reproduce

When play a game, open GameSwitcher, open another game

### Provide any additional context or information that may help us investigate your issue

[image]

Zazzago, 2025-04-24:
while if I use the 4.3.1-1 firmware, the glitch is not there, the only problem is that GameSwitcher always runs at 640x480 in any case with consequent "dirty" image if you use the overlay filters on MMV4

agycbkb, 2025-04-27:
As a workaound, my best guess for this issue is to still use 640x480 on MMV4 and give up on 752x560 resolution altogether, it's not ideal but if the Onion devs won't support it, it's the easiest solution. I'm still trying to figure out how to edit the overlays to work for this setup. I'm still hoping someone else might do this.

Zazzago, 2025-04-27:
> As a workaround, my best guess for this issue is to still use 640x480 on MMV4 and give up on 752x560 resolution altogether, it's not ideal but if the Onion devs won't support it, it's the easiest solution. I'm still trying to figure out how to edit the overlays to work for this setup. I'm still hoping someone else might do this.

the best hardware solution would have been for v4 to mount a 640x480 screen like the "plus", because a higher resolution doesn't make any sense for such a small screen, and always keeping the 640x480 resolution ruins all the overlays.

however I found a workaround to remove the glitch:

in the folder of the "theme" you use, modify the "pop-bg.png" file with a 640x480 screen that is all black (or in any case of a single color).
edit the theme file "config.json" in the line:
"grid: {"
change "color": to "#202020" (or the color of the screen above).
now the pop-bg.png screen that appears when you open a game is an "all black" screen, and you won't see any glitches, this solution only makes sense using light themes because with dark themes you won't be able to read the writing

agycbkb, 2025-04-27:
I see, if it was MainUI issue there's no wonder why Onion devs won't touch it. Your solution certainly work, I tried it but I think my solution won't compromise any theme at all. When I got back on my PC, I would try to create overlays to test then post it, if I made it to work.

agycbkb, 2025-04-28:
I'm new to this, I just got my device like a week ago. Sorry for my ignorance but most of the overlays work fine if one would keep the 640x480 resolution. I just got accustomed with those that works on 752x560 and also I like them better, maybe because they work really well.

Zazzago, 2025-04-28:
If you don't have specific screen overlays for 752x560 resolution, you will see the dirty image with those designed for 480p (especially GB and GBC). if you have a 560p screen and run it at 480p you will still see dirty image even with 480p overlays, because the image is scaled to the screen, not native

Zazzago, 2026-01-19:
[image]

I have the possibility to try multiple operating systems on multiple microSD, and I noticed that that "graphic glitch" on my miyoo mini v4 only appears in the new "beta" versions, in the old "stable" version it isn't there (because the "loading screen" doesn't appear when you start a game from gameswitcher). the only problem is that with version 4.3.1 the resolution when switching from a game to gameswitcher goes from 720x560 to 640x480, creating graphic glitches if you use overlay filters, while with the latest betas when switching from the game to gameswitcher the resolution remains the same, not creating glitches. 
my question is: Is it possible to modify only this part of code or is it a very complicated job that would take a long time? because with 4.3.1 if this problem is solved it is perfect for miyoo mini v4 with zero glitches