OnionUI/Onion #1808: Pop-up message "auto load state"

Zazzago, 2025-04-29:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.4.0-beta2-f11a0d

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

it's mainly an aesthetic glitch, but since it wasn't there in all the old onion versions I'll point it out.
basically its an "auto load state" pop-up message, that appears at the bottom left of the screen, with the relative loading percentage, every time I start a game from the GameSwitcher.

### Steps to Reproduce

Open GameSwitcher and open a game

### Provide any additional context or information that may help us investigate your issue

I think its a "problem" of retroarch update

filterpaper, 2025-05-01:
You can try adding `notification_show_save_state = "false"` into the `RetroArch/.retroarch/retroarch.cfg` file.

Zazzago, 2025-05-01:
hi Thanks, but I can't find the command line in retroarch.cfg. and then I have another doubt, isn't it that by disabling by default, it also disappears when you do a quick save and load?
because in the previous onion stable version, it didn't appear at the beginning, but when you made a save the pop-up was there

Zazzago, 2025-05-01:
> You can try adding `notification_show_save_state = "false"` into the `RetroArch/.retroarch/retroarch.cfg` file.

Adding the line, the problem is not solved: 
At the beginning, however, the "auto loading save state" pop-up appears, only the message disappears when you save or load with the key combination (menu+L2 menu+R2)

volen23, 2025-05-10:
Thank you, that line removed the save state text messages

Aemiii91, 2026-01-18:
Should be fixed in latest beta, please test (reopen if necessary)

closed 2026-01-18 by Aemiii91

Zazzago, 2026-01-18:
confirm it has been fixed, thank you