OnionUI/Onion #1812: Issue Title

Goodingn, 2025-05-11:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0 beta2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

When starting a Rom/Core it forgets how may save states there have been and positions itself at savestate 0.

I can make many saves, use the menu button and resume the game and it will remember the last save state number.

However, if I power off and back on and resume a game, it will load from state 0, not the most recent save.

It worked fine beta 1.

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

Above

aloven, 2025-09-19:
Likely solved with the new RA build (see here https://www.reddit.com/r/MiyooMini/comments/1n51a2c/gameswitcher_patch_for_v440beta2/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

But it also requires that you don't have savestates from a previous RA that exceed your auto-save interval setting.  So if it's set to 10, you can't have any savestates higher than 9  (starting from 0)  because if RA sees them, it will cycle back to 0 every game boot.

The older RA build (15?) autoincremented indefinitely (maybe to 999) and deleted the old savestates, but the new one runs through a cycle starting at 0, so it doesn't expect to see indexed savestate files higher than your auto-save interval.

RdrSeraphim, 2026-01-18:
This should be fixed with [2f76efe](https://github.com/OnionUI/Onion/commit/2f76efe1cd5169956746a6dbeed6775d6c664273), as this was a RetroArch bug.