OnionUI/Onion #1813: Gameswitcher not switching games on start up - Beta2

agycbkb, 2025-05-13:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.4.0-beta2

### Miyoo Firmware Version

0202408061133

### Provide a clear and concise description of the issue

Gameswitcher not switching games on start up when using Tweaks->System->Startup->Auto-resume last game ON. Whenever I power off while playing a game, boot up and hit the menu button to run Gameswitcher then pick another game, it doesn't change the game and it loads the last game played before powering off. 

### Steps to Reproduce

On beta2, play two or more games. Power off while playing a game. Power back on, when the game resumes hit the menu button to run Gameswitcher. Pick another game, it will load the first game on the list and not the one you pick. This only happens on first run of Gameswitcher after booting up.

### Provide any additional context or information that may help us investigate your issue

This doesn't happen on earlier versions.

agycbkb, 2025-05-14:
In addition to this, if you set Tweaks->System->Startup->Start application <Game Switcher> along with  Tweaks->System->Startup->Auto-resume last game ON and try to reproduce the issue, the Gameswitcher will run again after you try to pick a game. So in this setup you will have to pick the same game twice after booting up.

aloven, 2025-08-21:
I used Claude to help fix this issue and it seems to be working.  well, rather than figure out why autoresumed games act differently, I have added logic to gameSwitcher to detect autoresumed games and gracefully shutdown RetroArch before starting the overlay.  It then behaves like it was launched from the OS and successfully can select any of the games right the first time. 

https://github.com/OnionUI/Onion/pull/1845

closed 2026-01-18 by Aemiii91