OnionUI/Onion #1815: Saving issues with GBA (gpSP) on latest beta

Josh5, 2025-05-16:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.4.0-beta2-f11f7a0d

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I am having an odd issue since updating to the latest beta build.

I am testing and able to reproduce this with Golden Sun and Pokemon FireRed.

I cannot save a game. The states seem to work fine, but I cannot save the game to a .srm file. The game will say that it is saved, and if I reset the ROM while retroarch is still running it loads that saved game. Even if I quit out with the GameSwitcher button (but not closing the GameSwitcher) then reload that state the game straight away, then that saved game is able to be loaded after resetting the ROM in retroarch.
However, if I quit the ROM, it will always reload the same game save from a while back.

This was a bit hard to describe so I created a video of me reproducing it with GoldenSun (link below).

I have reproduced this with a brand new SD card and a fresh install of OnionOS

### Steps to Reproduce

# Steps to reproduce 
1) Load the game.
2) Save the game.
3) Exit the game with the GameSwitcher button, then press _"B"_ to exit.
4) Select _"Y"_ to open the GLO
5) Select _"Reset Game"_
6) Select the save game.
7) (Will fail to load the correct save game)

# Steps to reproduce a working load
1) Start the game.
2) Save the game.
3) Exit the game with the _GameSwitcher_ button (but don't quit GameSwitcher).
4) Click the A button to reload that auto state from GameSwitcher.
5) Open Retroarch menu and reset the ROM.
6) Select the save game.
7) (Will succeed to load the correct save game)

### Provide any additional context or information that may help us investigate your issue

Here is a link to a video showing the issue
https://photos.app.goo.gl/gvWD8eX3JhK3Zj5BA

Josh5, 2025-05-16:
If this is a gpSP core issue, I appologise. I just thought that it was strange that it worked when I would exit the game to the GameSwitcher and then reload. That made me wonder if this was OnionOS related.

I also did not notice this issue on the old version of gpSP with the stable OnionOS build. So I will be downgrading back to that soon to test if it starts working again.

maj0r-zero, 2025-05-17:
I'm having the same issue, both on gpSP and mGBA. Like you said, using the in-game save function does not create a .srm file. Tested various Pokémon titles and Donkey Kong Country.

Josh5, 2025-05-18:
I also was able to get my .srm files working correctly in mGBA and gpSP again after downgrading back to stable.

I've created a second SDCard for dev beta builds. I'll build the latest master and then start playing around with everything to figure out how it all works. Then I'll see if I can track down what is causing this.

Josh5, 2025-05-18:
After some more tests, I found that whenever I launch a game (for either gpSP or mGBA core via the GLO and selecting the "Reset game" option, it would load the saves correctly. Then if I dont exit with the GameSwitcher but instead bring up the Retroarch menu while the game is running and select "Quit", it exits back to the Onion UI GBA menu  and if I bring up the GLO and select the "Reset game" option again, that newly saved game worked and the .srm file was saved to the correct directory.

Josh5, 2025-05-19:
It seems that maybe the core is not writing out the .srm until retroarch exits. If I exit with bringing up the retroarch quick menu and then selecting "Quit", there is a good chance that that .srm save is written to the SD Card and is then there the next time I start up the game.
Perhaps quitting by the function button quits retroarch before it can write out that .srm file?

I had another suspicion about mount namespaces, but I am not sure about that one yet. I need to do more testing.

Josh5, 2025-05-19:
OK, so I think I have figured something out.

1) Retroarch and the gpSP core is not writing out the save until retroarch closes the core. I believe Retroarch will keep this save in memory until it runs the "close content" process. This then writs that data to disk.
2) The[ `gs_overlay.h`](https://github.com/OnionUI/Onion/blob/cff07dfebe7b7ea9e010bc21b96215f4ac950676/src/gameSwitcher/gs_overlay.h#L150) in GameSwitcher is killing retroarch with `killall -9 retroarch`.

Because of this, the states are written when you quit the game with the function button, but then the GameSwitcher kills the retroarch process before it has time to complete the close content process and quit.
This at least explains most of what I have been seeing.

Josh5, 2025-05-20:
> Sounds like the same issue as #1810. Nice find!

Ah, true. Yea, one of the issues I was seeing was .gba files being created next to the .zip roms. This must be retroarch, extracting them, and then not cleaning it up.

I've been using my build with the PR changeset applied for the past few days without issues.