OnionUI/Onion #1827: [CloudSync] could not begin

dpelevin, 2025-06-08:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion-v4.4.0-beta2

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I am trying to setup Cloud Sync across Miyoo Mini Plus, PC (Steam version of RetroArch) and Android version (https://buildbot.libretro.com/nightly/android/2025-06-07-RetroArch_ra32.apk). Yandex Disk is used as a WebDAV storage (https://disk.yandex.com/).

The same Cloud Sync settings are used with all three devices:
> webdav_password = "<app_password>"
> webdav_url = "https://webdav.yandex.com/RetroArch"
> webdav_username = "<username>"

I am succeed with syncing across Steam and Android. With Miyoo Mini Plus I am getting the following error:

> Cloud Sync at Miyoo Mini Plus "could not begin".

[retroarch_onionos_440_beta2.log](https://github.com/user-attachments/files/20644998/retroarch_onionos_440_beta2.log)
[retroarch_steam.log](https://github.com/user-attachments/files/20645136/retroarch_steam.log)

### Steps to Reproduce

1) Create Yandex ID account https://id.yandex.com/
2) Сreate App Password for Miyoo Mini Plus https://id.yandex.com/security/app-passwords 
3) Turn on Cloud Sync and specify the following settings:

> webdav_password = "<app_password>"
> webdav_url = "https://webdav.yandex.com/RetroArch"
> webdav_username = "<username>"

4) Launch RetroArch with Apps -> RetroArch
5) Observe the following message:

> 100%: Cloud Sync failed

### Provide any additional context or information that may help us investigate your issue

[retroarch_onionos_440_beta2.log](https://github.com/user-attachments/files/20644998/retroarch_onionos_440_beta2.log)
[retroarch_steam.log](https://github.com/user-attachments/files/20645136/retroarch_steam.log)

erikhoggard, 2025-07-29:
Hey, I'm having the same issue here. Did you ever figure it out?

h3xp, 2025-08-29:
I have the same issue with nextcloud.