OnionUI/Onion #1852: Switching to and from Guest Mode causes time travel for devices with an RTC

ThatStargirl, 2025-09-28:
### Checklist

- [x] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [x] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [x] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.4.0-beta2-f11f7a0d

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When switching between the guest and main profile, the clock is set back to whatever time the last profile switch took place at. 

### Steps to Reproduce

Using a Miyoo Mini/Miyoo Mini Plus with an RTC:

1. Switch into guest mode.
2. Wait
3. Switch out of guest mode; check and time will be out of sync.

### Provide any additional context or information that may help us investigate your issue

Seems to be caused by an intended feature of Guest Mode for devices without an RTC: storing and restoring the time on switching profiles.

fengfrw, 2025-09-28:
Try unsaveTime and loadTime in launch.sh for devices with RTC.
My mini can't be tested without rtc.

ThatStargirl, 2025-09-28:
Yep! I fixed it on my device earlier and wrote the bug report so I could submit a pull request with the change 😅 it does exactly that: skips saveTime.sh and loadTime.sh if the device has an RTC.